clockworkpi/DevTerm #1: keyboard mini

zerotrzy, 2020-11-24:
In my opinion this keyboard will be great version https://klawiatura.files.wordpress.com/2020/08/kl.jpg
big potenciometer, and moving status/menu bar to keyboard. 
i3wm + led keys (which virtual screen use)
change + key (not necessary shift to use +) etc.

Is possible buy version without printer but with battery for 24h working time?

carlsmith, 2021-01-08:
The link is dead.

trollsid, 2021-02-07:
https://www.pine64.org/wp-content/uploads/2020/07/PinePhonekeyboard2.jpg
Look like n950 keyboard or enhanced version of n900

dangpzanco, 2022-04-21:
I don't quite understand the motivation of this issue (maybe it should be closed?), but since the link was dead I got it from web archive:

https://web.archive.org/web/20201121135805/https://klawiatura.files.wordpress.com/2020/08/kl.jpg

This keyboard features a very unusual design: the idea of the potentiometer (the 2 circles on top right corner) is very interesting from the perspective of using the DevTerm as a retro digital synth, but the QWEȨRTπYUIOÓP ordering of keys is WEIRD.