clockworkpi/DevTerm #2: Bill of materials

blackketter, 2020-11-25:
Thanks for posting the main board schematic!  Unfortunately it seems to be missing part numbers.  Would it be possible to post a bill of materials with the component part numbers and the corresponding references that appear on the schematic?

tsacks, 2020-11-29:
This might not be worth much at the moment, but I think I might have figured out the ICs based on the function and pinouts.

U101- AXP221 PMIC
U201- RF Link K019-CW43-DW
U301- ST HDMI05-CL01F3
U302- Broadchip BCT3661B
U501- GL850G USB 2.0 hub chip

Godzil, 2021-03-29:
The BOM is not necessarily the issue, would be nice to have, but you can derive it from a schematic. What really miss is that all of the component values are missing (which include the part number)