clockworkpi/DevTerm #4: I'm keen to buy one, but it seems like it wouldn't work.

carlsmith, 2021-01-08:
Like many hobbyists, I have badly wanted something similar to DevTerm for years, and am gutted it looks like it's fatally flawed. My only real reservation is that the keyboard looks way too small for even a teenager to type on, so it'd be as frustrating as hell to edit code.

Do you envision grown ups coding on the DevTerm for even an hour or two? Is that even realistically possible?

carlsmith, 2021-01-08:
I can't speak for other hackers, but the value in something like this for me is in having a platform that is the same for every member of the community, like we had with the C64 or Beeb or whatever. There are many advantages; for example:

* Optimizing software to get the most out of the hardware is much easier if every user has the same CPU, GPU and memory.
* Responsive interfaces suck. Making graphical user interfaces is much easier (and much more fun) if you know the display resolution in advance (especially when using pixel art).
* Supporting a range of gamepads and keyboards, with different features, legends et cetera is much more complex than supporting exactly one of each.
* Writing docs is way easier (and the result is much nicer) if the author has the exact same system as the reader, especially when documenting lower level stuff like assembly or shaders.

I understand that you intend to support many hardware configurations with a range of operating systems, but that does not prevent a subset of the community opting to use the same setup and doing their own thing.

afritsch89, 2021-03-03:
Going off of eyeballing the USB port, it's small, but I've put in many hours on an atari portfolio, you would be surprised how usable a keyboard this size is.

But I guess it would definitely fall under a "to each their own" kinda thing...

StavroMueller, 2021-03-16:
I just plan on, honestly, using it with an external keyboard and then the internal one when I can't do that.

Godzil, 2021-03-29:
To be frank, here is not really a "customer service", so if you have any questions, direct them to CPi email, not in a GitHub issue.