clockworkpi/DevTerm #5: Please publish the DevTerm CM3 changes

Godzil, 2021-09-01:
Changes compared to the original Raspbian distribution

That include the changes done to the linux kernel if any, but considering that you use `devterm-kernel7.img` for the kernel name is it at least custom built, so the config file would be needed

This also include all the DTS overlay file that are included in the default CM3 image.

And any other relevant changes done.

It would be nice also to have a documentation on how the image was built. I think from the `issue.txt` in `/boot` you used pi-gen ( https://github.com/RPi-Distro/pi-gen ), what option were used to generate the image?

cuu, 2021-09-01:
sure thing, in couple of days I will make a doc for that
and I did not use pi-gen, I just use chroot to do the modifications

cuu, 2021-09-02:
# Devterm CM3 rpi os image

### Orginal image download url
https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-full.zip

#### Modification request list 
- [x] chromium downgrade to 84.0.4147.141 
- [x] screen rotation for xrandr,fbcon,lightdm
- [x] Remove the wallpaper and change the solid color #202020  
- [x] Install all rpi-related debs in the apt source 

#### Packages in apt source
1. devterm-audio-patch (check whether the 3.5 audio interface is plugged in, pull up a certain GPIO)
1. devterm-fan-temp-daemon-rpi (detect the temperature of rpi, raise the io drive fan, written in python)
1. devterm-kernel-rpi (modified kernel, 4.x series)
1. devterm-keyboard-firmware (keyboard firmware, just copy to /usr/local/bin, users need to manually flash it)
1. devterm-thermal-printer (printer program and scripts such as systemctl)
1. devterm-thermal-printer-cups (CUPS filter for thermal printer, and add thermal printer to cups, devterm printer can be seen in chrome) 

##### Add clockworkpi to /etc/hosts, then install devterm related apt, see https://github.com/clockworkpi/apt/tree/main/debian 
```
wget -O - https://raw.githubusercontent.com/clockworkpi/apt/main/debian/KEY.gpg | sudo apt-key add -

echo "deb https://raw.githubusercontent.com/clockworkpi/apt/main/debian/ stable main" | sudo tee -a /etc/apt/sources.list.d/clockworkpi.list
```

##### config xrandr 
```
sudo bash -c 'cat << EOF > etc/X11/Xsession.d/100custom_xrandr
xrandr --output DSI-1 --rotate right
EOF'
```

##### config lightdm
in `/etc/lightdm/lightdm.conf`
```
greeter-setup-script=/etc/lightdm/setup.sh
```
*setup.sh*
```
#!/bin/bash
xrandr --output DSI-1 --rotate right
exit 0
```
`sudo chmod +x etc/lightdm/setup.sh`

##### Modify /etc/dphys-swapfile
```
CONF_SWAPSIZE=512
```

##### Change the default wallpaper
The following files changed 

* /etc/xdg/pcmanfm/LXDE-pi/desktop-items-0.conf 
* /etc/xdg/pcmanfm/LXDE-pi/desktop-items-1.conf 
* /etc/lightdm/pi-greeter.conf 
* /home/pi/.config/pcmanfm/LXDE-pi/desktop-items-0.conf 
* 
Delete wallpaper address temple.jpg
Modify the desktop_bg color to #202020 

#### config.txt

Add dtparam=audio=on to solve the problem of no sound card under hdmi

##### /etc/hostname to clockworkpi

##### /boot/cmdline.txt ,add fbcon=rotate:1, remove quiet

voronoipotato, 2022-01-31:
Is there a timeline for the a04 imaging instructions? I presume it's going to be pretty similar to the A06 ones?

cuu, 2022-01-31:
@voronoipotato 
https://github.com/clockworkpi/DevTerm/wiki/Create-DevTerm-A04-OS-image-from-scratch

very similar to A06 image creation process

kpeeem, 2022-06-26:
+ Please publish the DevTerm CM3 too 🙏