clockworkpi/DevTerm #10: A06 Kernel gets replaced after build

reostat, 2021-11-27:
I'm following along the instructions described in `Create DevTerm A06 OS image from scratch` wiki page and there's something I don't quite understand. It seems that first we carefully apply kernel and u-boot patches and then build the image that's going to contain these patched kernel and u-boot; okay, makes sense so far. However, further down the lines in chroot we execute this
```bash
sudo apt remove linux-image*
sudo apt install devterm-kernel-current-cpi-a06 devterm-fan-daemon-cpi-a06 devterm-thermal-printer devterm-thermal-printer-cups devterm-wiringpi-cpi  devterm-first-start-a06
```
which to me looks like we completely remove our carefully patched and compiled kernel - perhaps of the exact version we wanted - and replace it with a precompiled binary from DevTerm repository.

Why do we do that?

cuu, 2021-11-27:
good point
when I try to build the a06 os image ,there is a step , I combined the build linux-dto* and linux-image* to became devterm-kernel-current-cpi-a06 
the reason is if not doing this , apt-get upgrade will cover the linux-dto,linux-image*, which will cause boot failed 
so to keep linux kernel in safe , I made devterm-kernel-current-cpi-a06 

but for the wiki, I am not sure how to explain this, you got the point right,it seems a little wired

reostat, 2021-11-27:
I see. There's a way to prevent updates of selected packages (including the kernel) in apt-based systems using `apt-mark hold <pkgname>`, perhaps it'd be a better way.

In the meantime, could you please publish the procedure you use to make `devterm-kernel-current-cpi-a06` so if somebody wants to add their own patches they can build the kernel themselves?

cuu, 2021-11-27:
first use armbian build system to build image 
after image done 
unzip linux-dtb-current-rockchip64, linux-image-current-rockchip64
and then combine all files ,all the postinst, preinst,prerm,postrm 
to be one deb file

closed 2021-11-28 by reostat

sakunamary, 2022-10-20:
> first use armbian build system to build image after image done unzip linux-dtb-current-rockchip64, linux-image-current-rockchip64 and then combine all files ,all the postinst, preinst,prerm,postrm to be one deb file

Sorry , I don't know  what  kinds of commands I should do to combine  postinst ,preinst,.... It 's mean put all file under one folder and then package all files into deb ; OR copy and paste then content into ONE  postinst ,preinst ... 
Thanks!
不好意思，我太理解combine 这个动作,在两个deb文件中的postinst preinst 文件解压缩完了之后，是放在一个新的文件夹里面（这样不会覆盖文件的问题么？），还是说将两个postinst 中的内容合并在一个新的postinst 中？
谢谢！