clockworkpi/DevTerm #11: `kernel-004-panel.patch` fails to apply

reostat, 2021-11-29:
While building A06 image according to instructions in `Create DevTerm A06 OS image from scratch` wiki, I noticed that `kernel-004-panel.patch` fails.

In order to stop the build process right at the moment it happens, run compile script with the following parameter:
```shell
sudo ./compile.sh EXIT_PATCHING_ERROR="yes"
```
Here's the output that I get:
```
... previous lines skipped
[ o.k. ] * [l][c] general-possibility-of-disabling-rk808-rtc.patch 
[ o.k. ] * [l][c] general-rk808-configrable-switch-voltage-steps.patch 
[ o.k. ] * [l][c] general-rkvdec_nv15.patch 
[ o.k. ] * [l][c] general-rockchip-overlays.patch 
[ o.k. ] * [l][c] general-rt5651-add-mclk.patch 
[ o.k. ] * [u][c] kernel-001-a06-dts.patch 
[ o.k. ] * [u][c] kernel-002-power.patch 
[ o.k. ] * [u][c] kernel-003-audio.patch 
[ warn ] * [u][c] kernel-004-panel.patch [ failed ]
[ error ] ERROR in function process_patch_file [ compilation.sh:960 ]
[ error ] Aborting due to [ EXIT_PATCHING_ERROR ]
[ o.k. ] Process terminated
```

reostat, 2021-11-30:
Alright, I dug it deeper and here's what panned out.

`Code/patch/armbian_build_a06/patch/kernel-004-panel.patch` introduces changes to the following files:
```
modify drivers/gpu/drm/panel/Kconfig
modify drivers/gpu/drm/panel/Makefile
create drivers/gpu/drm/panel/panel-cwd686.c
modify drivers/gpu/drm/rockchip/dw-mipi-dsi-rockchip.c
```
If applied as separate patches, all but the last one stick just fine. Further investigation shows that the changes in the last one are indeed already present in the latest v5.10.y kernel version; the changes were introduced - nearly verbatim - by commit id [43c2de10](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=43c2de10) of _May 28th, 2021 (2021-05-28)_. Removing changes to `dw-mipi-dsi-rockchip.c` file from the patch make it work again.

Now, this is probably not critical since AFAIK when patch fails it fails hunk by hunk, not all together; that is, in this case all the relevant parts would still get applied and the panel is going to work with this kernel anyway. However, as Linux kernel evolves there could be other cases when critical parts of patches might get rejected during the build thus producing an image that either doesn't work or can even potentially damage the hardware. After all, by default Armbian build reports failed patches at `warn` level and doesn't stop the build; locating those fails after the fact requires sifting through the build logs.

In order to prevent this from happening, I suggest doing the following.

1. Enable `EXIT_PATCHING_ERROR="yes"` by default so the build fails early should any of the patches fail
2. Fix u-boot and kernel versions to the last ones known to compile without errors with current config, e.g. using `KERNELBRANCH` and 'BOOTBRANCH` flags. Changing these versions should be recommended for advanced users only.
3. In addition to that, it might be a good idea to publish Armbian version the configs and instructions were verified against.

Together it should provide stable enough foundation for reproducible builds.

@cuu please let me know what you think. I'm working on a more Armbian-ish and less manual way to build an image in a separate fork; if the above sounds reasonable I will include it into configuration.

cuu, 2021-12-04:
totally agreed
and learned a lot from you

right now we are dealing the A04 kernel for armbian 
and if we done the A04 kernel ,we will try to   update the a06 patches and test
thanks very much

closed 2021-12-05 by reostat