clockworkpi/DevTerm #26: Ascii cinema for image builds

NeccoNeko, 2022-04-18:
Could you create Asciicinema recordings (or maybe even just plain YouTube recordings) of the image build process?

https://github.com/clockworkpi/DevTerm/wiki/Create-DevTerm-A06-OS-image-from-scratch

The wiki document lacks some information, which makes following the process difficult.

NeccoNeko, 2022-04-20:
Looking for full instructions, end to end, on building an image for my A06. The wiki document I linked is missing information and steps on building the base image, it errors out during compile.sh. 

I'm hoping, at the very least, to just get a recording of a beginning to end build so I can put together a process for automating it. 

If, instead, a working wiki document can be provided that would work as well. I just figured that a screen recording would be easier to provide.

cuu, 2022-04-20:
>  it errors out during compile.sh.

what  errors?

could you paste the output of the errors here ?

and also make sure to do the git hard reset step

> git reset --hard 43d179914ae9e1ebb5d72315d9f9f68f5fb3e330

the armbian building system is always updating,so errors might happen

NeccoNeko, 2022-04-20:
> what errors? ... could you paste the output of the errors here ?

You can see an example of errors on the ClockworkPi Forum:

https://forum.clockworkpi.com/t/does-anyone-have-a-dockerfile-for-building-the-devterm-image/8325/5

> and also make sure to do the git hard reset step

I did.  Here are the exact steps I performed (based on information from the wiki article and trial and error):

1. Installed ubuntu:21.04
2. Performed `apt update`
3. Installed the following packages
    * `apt install -y git dialog psmisc acl uuid-runtime curl gnupg gawk apt-utils lsb-release`
4. Checked out the required repositories:
    * `git clone https://github.com/armbian/build.git`
    * `git clone https://github.com/clockworkpi/DevTerm.git`
5. Performed a reset to the specified commit for the armbian build repository:
    * `cd build ; git reset --hard 43d179914ae9e1ebb5d72315d9f9f68f5fb3e330`
6. Created the required userpatches directory (this step was missing)
    * `mkdir -p userpatches/kernel/rockchip64-current/`
    * `mkdir -p userpatches/u-boot/u-boot-rockchip64-mainline/`
7. Applied the armbian.patch file from the DevTerm repository:
    * `git apply ../DevTerm/Code/patch/armbian_build_a06/patch/armbian.patch`
8. Copied the user patches into place:
    * `cp ../DevTerm/Code/patch/armbian_build_a06/patch/kernel*.patch userpatches/kernel/rockchip64-current/`
    * `cp ../DevTerm/Code/patch/armbian_build_a06/patch/uboot*.patch userpatches/u-boot/u-boot-rockchip64-mainline/`
9. Copied the config file into place:
    * `cp ../DevTerm/Code/patch/armbian_build_a06/patch/clockworkpi-a06.conf config/boards/`
10. Then attempted the armbian build process:
    * `./compile.sh`

No information was provided as to what to select when executing the build script.  A UI pops up asking questions, so I've had to guess my way through it and so far have ended up with the following so I don't have to interact with the UI to set options:

    BOARD=clockworkpi-a06
    # I've tried both 'legacy' and 'current' for the following:
    BRANCH=legacy   
    RELEASE=21.04
    # Left the following blank to keep things simple
    DESKTOP_APPGROUPS_SELECTED=
    DESKTOP_ENVIRONMENT=xfce
    DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base
    BUILD_MINIMAL=no
    BUILD_DESKTOP=yes
    KERNEL_ONLY=no
    KERNEL_CONFIGURE=no
    PROGRESS_LOG_TO_FILE=yes
    NO_APT_CACHER=yes

The image fails to build, specifically when trying to build the patches copied over.  I'll dig up the latest failed logs tomorrow to share (getting late here).

In the mean time, if you can review the steps I performed above and provide feedback/corrections that would be awesome.  

Or, if it wouldn't be too much to ask, a screen recording of the build from beginning to end should help in addressing the above, and any other possible gaps in the documentation.

sboger, 2022-04-20:
I have also attempted the build on a Linux Mint 20.3 laptop using the docker method. Same exact steps as above, but I run:
./compile.sh docker

The build fails at the kernel level. The kernel patches in /userpatches fail to apply.

sboger, 2022-04-20:
Here's a complete log of a failed build:

Log:
[devterm-build.log](https://github.com/clockworkpi/DevTerm/files/8525357/devterm-build.log)

Commands:
```
git clone https://github.com/armbian/build.git
git clone https://github.com/clockworkpi/DevTerm.git
cd build
git reset --hard 43d179914ae9e1ebb5d72315d9f9f68f5fb3e330
git apply ../DevTerm/Code/patch/armbian_build_a06/patch/armbian.patch
git apply ../DevTerm/Code/patch/armbian_build_a06/patch/armbian_mirror.patch
mkdir -p userpatches/kernel/rockchip64-current/
mkdir -p userpatches/u-boot/u-boot-rockchip64-mainline/
cp -f ../DevTerm/Code/patch/armbian_build_a06/patch/kernel*.patch userpatches/kernel/rockchip64-current/
cp -f ../DevTerm/Code/patch/armbian_build_a06/patch/uboot*.patch userpatches/u-boot/u-boot-rockchip64-mainline/
cp -f ../DevTerm/Code/patch/armbian_build_a06/patch/clockworkpi-a06.conf config/boards/
./compile.sh docker BOARD=clockworkpi-a06 BRANCH=current RELEASE=hirsute BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_ONLY=no KERNEL_CONFIGURE=no DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base COMPRESS_OUTPUTIMAGE=sha,gpg,img
```

sboger, 2022-04-20:
As a comparison, I tried compiling directly on the A06. Same as above, just ./compile.sh instead of ./compile.sh docker.

Same patch failures, same build error.

NeccoNeko, 2022-04-21:
@cuu have you seen the above?

NeccoNeko, 2022-04-21:
I've attached logs and an asciicinema cast file from my attempt to build Armbian for the A06, using the instructions from the wiki article.

[build_fail.cast.gz](https://github.com/clockworkpi/DevTerm/files/8535255/build_fail.cast.gz)
[debug_logs.tar.gz](https://github.com/clockworkpi/DevTerm/files/8535256/debug_logs.tar.gz)

NeccoNeko, 2022-04-21:
I've changed the title of this issue.

cuu, 2022-04-22:
after a little digging

first 
the kernel version has changed even we set the hard reset hash

what we use in a06 stock image is 5.10.60 and now it is 5.10.100 today 

so 
```
[ warn ] * [u][c] kernel-004-panel.patch [ failed ]
[ o.k. ] * [u][c] kernel-005-backlight.patch 
[ warn ] * [u][c] kernel-006-panel.patch [ failed ]
```
these two clockworkpi patches won't work even if the compile went through

second:
to check the compilation.log,this is real error output log

```
output/debug/compilation.log
````

`
	== u-boot make clockworkpi-a06-rk3399_defconfig ==

	== u-boot make u-boot-dtb.bin ==

.config:1793:warning: override: reassigning to symbol BOOTDELAY
tools/imx8mimage.c: In function ‘copy_file.constprop’:
tools/imx8mimage.c:291:7: warning: ‘write’ reading 5 bytes from a region of size 4 [-Wstringop-overflow=]
  291 |   if (write(ifd, (char *)&zero, 4 - tail) != 4 - tail) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from tools/imagetool.h:21,
                 from tools/imx8mimage.c:9:
/usr/include/unistd.h:367:16: note: in a call to function ‘write’ declared with attribute ‘read_only (2, 3)’
  367 | extern ssize_t write (int __fd, const void *__buf, size_t __n) __wur
      |                ^~~~~
tools/imx8image.c: In function ‘copy_file.constprop’:
tools/imx8image.c:414:7: warning: ‘write’ reading 5 bytes from a region of size 4 [-Wstringop-overflow=]
  414 |   if (write(ifd, (char *)&zero, 4 - tail) != 4 - tail) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./tools/imagetool.h:21,
                 from include/imx8image.h:13,
                 from tools/imx8image.c:8:
/usr/include/unistd.h:367:16: note: in a call to function ‘write’ declared with attribute ‘read_only (2, 3)’
  367 | extern ssize_t write (int __fd, const void *__buf, size_t __n) __wur
      |                ^~~~~
arch/arm/mach-rockchip/rk3399/rk3399.c: In function ‘board_debug_uart_init’:
arch/arm/mach-rockchip/rk3399/rk3399.c:125:36: warning: unused variable ‘gpio’ [-Wunused-variable]
  125 |  struct rockchip_gpio_regs * const gpio = (void *)GPIO0_BASE;
      |                                    ^~~~

	== kernel ==

scripts/Makefile.lib:8: 'always' is deprecated. Please use 'always-y' instead
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:22.11-27.6: Warning (spi_bus_reg): /fragment@1/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:36.11-41.6: Warning (spi_bus_reg): /fragment@2/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:50.11-55.6: Warning (spi_bus_reg): /fragment@3/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:64.11-69.6: Warning (spi_bus_reg): /fragment@4/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/rk3399-tinker-2.dts:401.23-422.4: Warning (i2c_bus_reg): /i2c@ff3c0000/vdd_cpu_b: I2C bus unit address format error, expected "60"
arch/arm64/boot/dts/rockchip/rk3399-tinker-2.dts:497.19-518.4: Warning (i2c_bus_reg): /i2c@ff3d0000/vdd_gpu: I2C bus unit address format error, expected "60"
drivers/video/fbdev/core/fbcon.c: In function ‘fbcon_scrolldelta’:
drivers/video/fbdev/core/fbcon.c:2998:16: error: ‘struct fbcon_display’ has no member named ‘scrollmode’; did you mean ‘yscroll’?
  switch (disp->scrollmode) {
                ^~~~~~~~~~
                yscroll
make[4]: *** [scripts/Makefile.build:280: drivers/video/fbdev/core/fbcon.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [scripts/Makefile.build:497: drivers/video/fbdev/core] Error 2
make[2]: *** [scripts/Makefile.build:497: drivers/video/fbdev] Error 2
make[1]: *** [scripts/Makefile.build:497: drivers/video] Error 2
make: *** [Makefile:1822: drivers] Error 2
make: *** Waiting for unfinished jobs....
`

after googled a bit

It may be is a armbian build system issue

looks like we need update the armbian build system to fix this kernel compilation issue

or manually fix other armbian patches for this newer kernel

cause this commit hash we set is only for kernel 5.10.60

and armbian continues to update kernel version 

so RIght now, you can just follow the wiki the produce the image as we want 

you need to fix it manually by yourself,update the patches to fit kernel updates

looks like manjaro is a better choice

NeccoNeko, 2022-04-22:
@cuu Thank you very, very much for reviewing and responding.

The wiki is incomplete.  It lacks instructions for creating directories to copy the patches into, plus the options to pick for compile.sh.  One of the options (board version) can't be set from the config UI that loads up, so it needs to be set manually at the CLI.

As for it being outdated, perhaps it can be updated, at the very least a warning that the instructions no longer work?  That way no one else spends time on something that won't work, unless they want to either 1) manually fix the kernel patch files to work with the latest kernel or 2) dig into the guts of the Armbian build script configuration files to set a fixed kernel version.

Again, thank you very much for your time on this.  It's greatly appreciated.

sboger, 2022-04-22:
Fixed. Create file build/userpatches/lib.config

lib.config:
`
KERNELBRANCH='tag:v5.10.60'
`

I've created pull-request https://github.com/clockworkpi/DevTerm/pull/28. It adds the file and improves the readme.

oscillot, 2022-04-23:
> 

omg i was right behind you I just didn't have `tag:`

littlecxm, 2022-05-12:
merged to armbian master now, you can try this https://github.com/armbian/build/commit/1f54bc33b8dda678de473d7d006a9a7f6ab0b86b