clockworkpi/DevTerm #31: CM3 shutdown doesn't kill +5V USB power.

markatlnk, 2022-04-27:
I don't believe it affects other CPU modules, just the Compute Module 3.

It looks like it shuts down, but the USB is still powered and it is pulling about 43mA from the batteries. After shutting down, if I take out the batteries and put them back in, current drops to 60uA closer to what I would expect.

Posting this on Discord, looks like other have also had this issue.

cuu, 2022-05-06:
how about sudo halt -p ?

markatlnk, 2022-05-06:
The power button doesn't close it down and shutdown -h now doesn't either.   The -P option is the default and shuts down every other pi I own. I only noticed this when I had an arduino plugged in and its lights were still blinking when I shut down the system. I had noticed several times when I pulled it out of my bag after a few days that the batteries were dead. Always thought something pushed the power button in my bag.  Shutdown is actually part of systemctl and I attempted a few options with that also. So far the only thing that works is momentary removal of batteries. Just a second or two will do it.

andrwj, 2022-09-16:
Then this problem is caused by hardware problem, can ClockworkPi show some guidelines to appy to CM3 DevTerm?

markatlnk, 2022-09-16:
Yes it is a hardware problem, but a minor one that can be resolved with changing the value of one resistor. These parts are really small so you do need a bit of experience with that. I teach Electrical Engineering and just happen to have a full SMT rework setup on my bench so it wasn't that difficult for me to do that. Because I wanted to lower the resistance of the value on the board, I chose to not remove the old part and just stick another part directly on top of it. It can be done with a microscope and a really fine tipped soldering iron. Make sure you use a drop of flux on there and if you use the no-clean type, you can just leave it. For those with better eyes than mine, you might get by with one of those magnifying glasses.  Hope that helps.  I think they used 0402 parts, which is way smaller than I typically work with, so I actually tacked an 0805 part off one side and a really tiny wire to connect the other side. Bit of a kludge, but it is still working just fine.  The device can sit for weeks without any battery drain.