clockworkpi/DevTerm #33: Failure to boot after upgrading R-01 OS

jw-0, 2022-06-01:
Hi, myself and others in the community have encountered an issue with the R-01 OS. If you run a full update of the OS packages with 'apt update && apt upgrade', once you attempt to reboot the OS will not start back up again and you will be left with a black screen.  The power light does come on down the bottom but nothing happens on screen and the OS does not boot back in and you cannot SSH in. The only remedy is to re-flash the SD card with the OS image available on [the site](http://dl.clockworkpi.com/). I tested it again after re-flashing SD card with the downloaded image and got the same result, it seems to be an issue with both the shipped OS on the SD card and the provided OS image. Further details: https://forum.clockworkpi.com/t/r-01-screen-wont-turn-on-again-after-shutdown/8527/6

I hope I am submitting the issue to the correct location, please let me know if elsewhere is more appropriate, thank you!

cuu, 2022-06-01:
sorry to hear that
but R01  use the devel branch of ubuntu(because we had to , for latest mesa to work with)

it has risks to upgrade, according your feed back ,it seems that the kernel has been misplaced

so mount the sd card(with R01 os)  on a PC (like Linux )

then go into the **boot partition** of the sd card

you will see 
```
extlinux.conf  
extlinux.conf.cpi.bak  
```
replace extlinux.conf  with extlinux.conf.cpi.bak

the **extlinux.conf.cpi.bak** is for this kind of situation

the apt upgrade will mess up the content of exlinux.conf

jw-0, 2022-06-01:
Thanks this worked, appreciate it

nzmichaelh, 2022-06-08:
FWIW, `/boot/exlinux/extlinux.conf` is updated by u-boot-update. It looks like you can turn off automatic updating by adding `U_BOOT_UPDATE=false` to `/etc/default/u-boot`.

cuu, 2022-06-08:
> FWIW, `/boot/exlinux/extlinux.conf` is updated by u-boot-update. It looks like you can turn off automatic updating by adding `U_BOOT_UPDATE=false` to `/etc/default/u-boot`.

great!

andypiper, 2022-06-09:
Could you say more about the Mesa issue, please? I'd like to be able to install libsdl2-dev but that seems to have a conflict caused by the version of Mesa (or something else in the graphics stack). 

What's the specific change to `extlinux.conf` that causes the boot failure?

thomasdstewart, 2022-06-09:
The official image leaves the Ubuntu RISC V kernels installed and after they get upgraded u-boot-update runs, which overwrites the extlinux.conf. Much like on a x86 system installing a new kernel updates the grub config file. The u-boot-update run makes a guess at which kernels you want to use, makes a menu to let you pick them all (you don't see the menu unless you see u-boot output in the debug uart) of course it assumes you want the latest version of linux, which does not boot. If you look at the content of the extlinux.conf after a u-boot-update run and the extlinux.conf.cpi.bak you can see they are completely different, as u-boot does not know about any of the custom allwinner d1 options. Probably the best quick option is setting that U_BOOT_UPDATE=false. The correct way would be to package the devterm r-01 kernel properly and understand how to integrate with u-boot-update more gracefully.