clockworkpi/DevTerm #39: Would this module work in a DevTerm?

rog77, 2022-07-03:
https://github.com/wuxx/icesugar-pro

Not sure of the connector compatability... 

The idea would be to use something like Litex to generate SoC with appropriate peripheral controllers.

Sorry if this is off topic.

cuu, 2022-07-10:
One thing I can confirm is this board can be plugged into DT
But I am not sure it can run under DT 
if all the GPIO is like as same as CM3, then there is a good chance that it will work

rog77, 2022-07-15:
https://github.com/wuxx/Colorlight-FPGA-Projects#ddr2-sodimm-200p

I compared the CM pinout to the above, and alas it it not the same. Thanks for your response - it seemed like an idea, but it looks like it is not to be.

closed 2022-07-15 by rog77