clockworkpi/DevTerm #40: Fan not working on CM4 64bit version

btyh17mxy, 2022-08-19:
I have change the MAX_TEMP to 60 in  /usr/local/bin/temp_fan_daemon.py file and restarted it, then got cpu temp over 70 but the fan still not running.

andypiper, 2022-09-04:
There are a number of issues here:

- The service still wants to use Python 2, which is obsolete (this is also unnecessary since the script has no Python 2 dependency)
- `/usr/local/bin/temp_fan_daemon.py` refers to `/opt/vc/bin/vcgencmd` which does not exist in Bullseye (this should be `/usr/bin/vcgencmd`
- It assumes that the fan is at GPIO 17, but I've tried manually switching GPIO 17 on and off and it has no effect on the fan.
 - the [wiki seems to say that the GPIO map is different on the CM4](https://github.com/clockworkpi/DevTerm/wiki/Devterm-CM4-GPIO-map) but it does not list "old" 17 so it is not clear whether this should still be used. 

At the moment, the DevTerm fan package will simply not work on the CM4.

andypiper, 2022-09-04:
This package should not be required if Raspberry Pi OS is being used - fan control is part of `raspi-config`. If we know the GPIO to use this can do the work without a Python script sleeping in the background.

cuu, 2022-09-05:
thanks 
the package  has upgraded

```
sudo apt update
sudo apt install devterm-fan-temp-daemon-cm4 -y
```

andypiper, 2022-09-06:
I've now removed the `devterm-fan-temp-daemon-cm4` package. I'm using the built-in Raspberry Pi OS settings to manage the fan and thermals:

In `raspi-config` -> Option 4 Performance -> Option 4 Fan
Yes, you want the fan control.
Fan on GPIO 17
Your choice of temperature threshold (minimum value is 60)

Then reboot.

andypiper, 2022-09-06:
@btyh17mxy you can try this:

- switch off the fan control service temporarily (to confirm that the GPIO works)
 - `systemctl stop devterm-fan-temp-daemon-cm4`

- check GPIO 17
 - `sudo gpio readall`
 - 17 should be set to OUT, Low (probably)
 - `sudo gpio toggle 17` will then toggle the power / set it to High, which should turn the fan on.

That should confirm whether the fan works in general.

btyh17mxy, 2022-09-07:
@andypiper  according to [wiki as you mentioned](https://github.com/clockworkpi/DevTerm/wiki/Devterm-CM4-GPIO-map) , the gpio pin for fan on CM4 should be 17. But the [temp_fan_daemon.py](https://github.com/clockworkpi/DevTerm/blob/main/Code/devterm_fan_daemon_rpi/temp_fan_daemon.py) seems got it wrong (still using 41).

so, I got the fan back to work by modify the `/usr/local/bin/temp_fan_daemon.py` file change gpio from 41 to 17 and `systemctl restart devterm-fan-temp-daemon`.

andypiper, 2022-09-07:
They fixed this in the -cm4 version, that script is from the -rpi version.

You do not actually need this extra code running (it wakes up every second to check the temperature) - you can use the built-in settings instead.