clockworkpi/DevTerm #42: GPIO for specific DevTerm hardware with CM4

andypiper, 2022-09-04:
The fan control script #40 and the audio patch script (`/usr/local/bin/audio_3.5_patch.py`) rely on specific GPIO pins for control of the DevTerm hardware.

What GPIO pins are used for the Fan, and the Audio Jack? What other GPIO pins map to specific hardware features of the DevTerm?

cuu, 2022-09-05:
Fan GPIO is 17
Audio Jack is 11 and 10

andypiper, 2022-09-06:
I'm unable to get the fan to spin up on GPIO 17. I've tried directly setting GPIO 17 as an output and then setting it to high/1 (using the `gpio` command) but the fan doesn't spin up. I'll try again later, but is there anything that could be missing here?

andypiper, 2022-09-06:
OK. Confirmed that the fan is working on GPIO 17 - thank you.