clockworkpi/DevTerm #43: CM4 packages cause Python 2 to be installed

andypiper, 2022-09-04:
The packages `devterm-audio-patch` and `devterm-fan-temp-daemon-cm4` have an (invalid / incorrect) dependency on python2, which causes several obsolete Python packages to be installed.

These should be edited to change `/usr/bin/python2` to `/usr/bin/python3` in the `.service` files, and have the python2 dependency updated to python3 in the the APT/DEB packaging.

cuu, 2022-09-05:
done

andypiper, 2022-09-05:
Thanks for fixing the packaging. Can you also check issue #40 and #42 and confirm the GPIO for the fan when using the CM4. The fan-temp-daemon is probably not needed but we would need to know which GPIO switches it on and off.

cuu, 2022-09-05:
it is the 17
and with upgraded  fan-temp-daemon,i've checked that the fan can work , I just set MAX_TEMP to 50 in py code,then the fan runs

if you can not run the fan with 17 GPIO,maybe is because your wiringpi library is not for  devterm cm4

try the devterm cm4 wiringpi library

andypiper, 2022-09-06:
OK thank you I will recheck this.

andypiper, 2022-09-06:
OK cool - confirmed the fan is working here.

closed 2022-09-06 by andypiper

fei20th, 2024-03-28:
> The packages `devterm-audio-patch` and `devterm-fan-temp-daemon-cm4` have an (invalid / incorrect) dependency on python2, which causes several obsolete Python packages to be installed.
>
> These should [...]

where are those .service files located?